Wind-River/vxworks7-ros2-build #3: Simple CMake example: fseeko64 not found

rcwoolley, 2020-03-16:
The instructions for "Build a simple CMake based OSS project" don't work anymore.

Recent changes to tinyxml2 introduce a #ifdef switch in tinyxml2.cpp that defines TIXML_FSEEK to fseeko64 as a default for x86_64.

Since fseeko64 doesn't exist in VxWorks it fails to compile.

A temporary workaround for the documentation is to instruct the user to checkout 8c8293ba8969a46947606a93ff0cb5a083aab47a in the tinyxml2 tree which predates the change.

razr, 2020-03-17:
Thanks @rcwoolley.
 
I have left a comment here:
https://github.com/leethomason/tinyxml2/issues/786#issuecomment-600164731

as a possible workaround:
cmake -E env CXXFLAGS="-Dfseeko64=fseek -Dftello64=ftell" cmake .. -DCMAKE_MODULE_PATH=$TOP_BUILDDIR/buildspecs/cmake -DCMAKE_TOOLCHAIN_FILE=$TOP_BUILDDIR/buildspecs/cmake/rtp.cmake

razr, 2020-05-06:
it was fixed by tinyxml2 project
https://github.com/leethomason/tinyxml2/commit/99db0247d6332bb9b501f9f2972c0332ed8831b6
I'll keep a README unchanged

closed 2020-05-06 by razr