Wind-River/vxworks7-ros2-build #7: Shared object not found

pablogs9, 2020-05-27:
Hello, I have been following [VxWorks 7 SDK for Raspberry Pi 4](https://labs.windriver.com/downloads/wrsdk-vxworks7-docs/README-raspberrypi4b.html) instructions and [this repo instructions](https://github.com/Wind-River/vxworks7-ros2-build/tree/dashing-release-SR0640#build-vxworks-7-and-ros2) in order to run some ROS2 demos on VxWorks 7 on a RPi 4.

I mount the `export` folder by FTP (like in the SDK for RPi4 instructions) and I try to execute the same commands you expose [here](https://github.com/Wind-River/vxworks7-ros2-build/tree/dashing-release-SR0640#run-ros2-examples). So, I have:

```
-> netDevCreate("wrs","192.168.1.108",1)                                                                                                                                                                           
value = 0 = 0x0                                                                                                                                                                                                    
-> cmd                                                                                                                                                                                                             
[vxWorks *]# cd wrs
[vxWorks *]# ls
include
lib
llvm
[vxWorks *]# set env LD_LIBRARY_PATH="/wrs/lib"
[vxWorks *]# getenv LD_LIBRARY_PATH
getenv: LD_LIBRARY_PATH = /wrs/lib
[vxWorks *]# cd llvm/bin
[vxWorks *]# rtp exec -u 0x20000 timer_lambda.vxe
Launching process 'timer_lambda.vxe' ...
Process 'timer_lambda.vxe' (process Id = 0xffff8000007dae30) launched.
Shared object "librclcpp.so" not found
```
`librclcpp.so` is actually in `/wrs/lin` folder.

razr, 2020-05-27:
Hello @pablogs9, could you please deploy the content of the export/root/ dir on your SD Card and run it from there? You don't need an FTP connection for it.

pablogs9, 2020-05-28:
Hello @razr, I have copied the folder `export` in the root of the SD card, but now I have this scenario:

```
[vxWorks *]# pwd                      
/sd0a/export
[vxWorks *]# set env LD_LIBRARY_PATH="/sd0a/export/lib/"
[vxWorks *]# ls  
. 
.. 
lib 
include 
llvm 
[vxWorks *]# cd llvm/bin
[vxWorks *]# pwd
/sd0a/export/llvm/bin
[vxWorks *]# rtp exec -u 0x20000 timer_lambda.vxe
Launching process 'timer_lambda.vxe' ...
Process 'timer_lambda.vxe' (process Id = 0xffff800000607000) launched.
terminating with uncaught exception of type rclcpp::exceptions::RCLError: failed to initialized rcl init options: failed to load shared library of rmw implementation. Exception: , at /work/build/ros2/ros2_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:161, at 5
0xffff800000602910 (iTimer_lambda): RTP 0xffff800000607000 has been deleted due to signal 6.
```

By the way, I also have had errors while copying some files to the FAT32 partition because it does not support symlinks.

danmilea, 2020-05-28:
Hi @pablogs9, when deploying content to the SD card use cp with the -L flag (-L, --dereference -> always follow symbolic links in SOURCE). This will solve the symlink related issues.

pablogs9, 2020-05-29:
Working, the deference flag solved the not found error.

Thank you so much @razr @danmilea.

closed 2020-05-29 by pablogs9