Wind-River/vxworks7-ros2-build #8: Error when building RCLCPP

pablogs9, 2020-09-21:
Hello, I have been following [VxWorks 7 SDK for Raspberry Pi 4](https://labs.windriver.com/downloads/wrsdk-vxworks7-docs/README-raspberrypi4b.html) instructions and [this repo instructions](https://github.com/Wind-River/vxworks7-ros2-build/tree/dashing-release-SR0640#build-vxworks-7-and-ros2) in order to run some ROS2 demos on VxWorks 7 on a RPi 4.

I get the following error when building `rclcpp`:

```bash
--- stderr: rclcpp                                                                                                              
CMake Warning:
  Manually-specified variables were not used by the project:

    PYTHON_SOABI

In file included from /work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/client.cpp:15:
In file included from /work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp:31:
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:93:2: error: "Unsupported C++ compiler / standard library"
#error "Unsupported C++ compiler / standard library"
 ^
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:95:3: error: expected unqualified-id
  : function_traits<ReturnTypeT(Args ...)>
  ^
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:110:2: error: "Unsupported C++ compiler / standard library"
#error "Unsupported C++ compiler / standard library"
 ^
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:112:3: error: expected unqualified-id
  : function_traits<ReturnTypeT(Args ...)>
  ^
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:124:2: error: "Unsupported C++ compiler / standard library"
#error "Unsupported C++ compiler / standard library"
 ^
/work/vxworks7-ros2-build/build/ros2/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:126:3: error: expected unqualified-id
  : function_traits<ReturnTypeT(Args ...)>
  ^
```

Can I have some support with errors? Thank you so much!

sdoerr20, 2020-09-22:
Currently, I am facing the same issue for the RPI 3B+ SDK. As I successfully build the libraries before at the end of June using the Docker Build infrastructure, I was surprised by the fact that recently the docker build fails involving the ROS2 rclcpp package. I have tested the following versions with nealry the same outcome:

| vxworks-ros2-build branch | wrsdk RPI3 version | Outcome |
| ------------------------- | ------------------ | ------- |
| dashing-releaseSR640   | 1.4 | rclcpp failed |
| SR0650                 | 1.5 | Make-Target fails because of missing rule |
| master                 | 1.5 | Fails as dummy.c is missing |
| master                 | 1.4 | rclcpp failed |
| dashing-release        | 1.5 | fails to configure asio |
| dashing-release        | 1.4 | rclcpp failed |
| dashing-release-python | 1.4 | rclcpp failed |

Is there a supported set of fixed combinations of different versions? For further debugging purposes I attached my |trace](https://github.com/Wind-River/vxworks7-ros2-build/files/5259602/issue_trace.txt) as well.

razr, 2020-09-22:
Hello @sdoerr20, thanks for reporting it. It shall be possible to build an RPI3 image with ROS2 against the SR0640 SDK which is https://labs.windriver.com/downloads/wrsdk-vxworks7-raspberrypi3b-1.4.tar.bz2
Let me look into it.

razr, 2020-09-22:
Hello @pablogs9, thanks for reporting it. The RPI4 SDK for ROS2 is also the SR0640 one -https://labs.windriver.com/downloads/wrsdk-vxworks7-raspberrypi4b-1.0.tar.bz2. Will look into it as well.

razr, 2020-09-23:
This issue shall be fixed now with https://github.com/Wind-River/vxworks7-layer-for-ros2/commit/81197b31d35cc55fe1fe1c413cc50c68564036fe
I have tested it on QEMU, please let me know if it works for you, that I can close it

pablogs9, 2020-09-24:
Thanks @razr for this update. Closing the issue.

closed 2020-09-24 by pablogs9