Wind-River/vxworks7-ros2-build #9: Help for debugging VxWorks error signal 11

pablogs9, 2020-09-23:
Hello, I'm using this package to build `micro-ROS-Agent`, the server part of micro-ROS (ROS 2 for microcontrollers).
We have solved compilation issues by patching some packages and tweking the build procedure in order not to use some broken ROS 2 packages (related to https://github.com/Wind-River/vxworks7-ros2-build/issues/8).

Once we achieve running this binary inside VxWorks in a Raspberry Pi 4 it seems to run correctly until a certain point (when ROS 2 datawriter are created).

The VxWorks output is:

```bash
-> cmd   
[vxWorks *]# set env LD_LIBRARY_PATH="/sd0a/export/root/lib/"
[vxWorks *]# cd /sd0a/export/root/llvm/bin
[vxWorks *]# rtp exec -u 0x2000000 micro_ros_agent.vxe "udp4 --port 8888 -v6"
Launching process 'micro_ros_agent.vxe' …
Process 'micro_ros_agent.vxe' (process Id = 0xffff800000d66560) launched.
Press CTRL+C to exit
0xffff8000010bcbe0 (t40): RTP 0xffff800000d66560 has been deleted due to signal 11.
```

The complete procedure of building and setting up can be found [here](https://docs.google.com/document/d/1G1hR4TwZkWrnwQO8E3KZSITtovbgPHlFiUXTxG8NmQc/edit?usp=sharing).

My question is: where can I find instructions to debug this `signal 11` error in VxWorks BSP for RPi 4? Is there any way of having a explicit trace of the error?

PD: I'm not sure if this is offtopic for this repo issues, if so, I will close it.

Thanks!

razr, 2020-09-24:
@pablogs9 thanks for providing patches. There are other two patches I need to apply and after that, I can build the whole project.
It runs on my side without a crash as

```
akholodn@razilix:~/github/Wind-River/vxworks7-ros2-build$ qemu-system-x86_64 -m 512M  -kernel $WIND_SDK_HOME/bsps/itl_generic_2_0_2_1/boot/vxWorks -net nic  -net tap,ifname=tap0,script=no,downscript=no -display none -serial stdio -monitor none -append "bootline:fs(0,0)host:vxWorks h=192.168.200.254 e=192.168.200.1 u=target pw=boot o=gei0" -usb -device usb-ehci,id=ehci  -device usb-storage,drive=fat32 -drive file=fat:rw:./export/root,id=fat32,format=raw,if=none
```

and then

```
-> cmd
[vxWorks *]# set env LD_LIBRARY_PATH="/bd0a/lib"
[vxWorks *]# cd /bd0a/llvm/bin/
[vxWorks *]# rtp exec -u 0x20000 micro_ros_agent.vxe "udp4 --port 8888 -v6"
Launching process 'micro_ros_agent.vxe' ...
Process 'micro_ros_agent.vxe' (process Id = 0xffff8000005b6ed0) launched.
Press CTRL+C to exit
```

I'll integrate your patches later on today.
As for debug you can refer to https://labs.windriver.com/downloads/wrsdk-vxworks7-docs/VxWorksSDK-ApplicationDeveloperGuide.html#run-and-debug-from-cli

pablogs9, 2020-09-24:
Hello @razr thank you very much for testing this. For me it works until a client tries to create a datawriter. In fact, I have found that the agent is able to create a participant, a topic and it crashes when creating a datawriter.

Have you tried with your current setup to connect a client to this agent?

razr, 2020-09-24:
no, I haven't, could you please provide more instructions on it? Probably just extend your [google doc](https://docs.google.com/document/d/1G1hR4TwZkWrnwQO8E3KZSITtovbgPHlFiUXTxG8NmQc/edit) ?

pablogs9, 2020-09-24:
For sure:

```bash
git clone https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build; cd build;
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON  
make -j 12
./examples/PublishHelloWorld/PublishHelloWorldClient [VxWorks machine IP] 8888
```

You also have them in the google doc.

pablogs9, 2020-09-24:
Regarding the QEMU command: 

```bash
sudo qemu-system-x86_64 -m 512M  -kernel /home/pgarrido/new_vxworks/wrsdk-vxworks7-qemu/bsps/itl_generic_2_0_2_1/boot/vxWorks -net nic  -net tap,ifname=tap0,script=no,downscript=no -display none -serial stdio -monitor none -append "bootline:fs(0,0)host:vxWorks h=192.168.200.254 e=192.168.200.1 u=target pw=boot o=gei0" -usb -device usb-ehci,id=ehci  -device usb-storage,drive=fat32 -drive file=fat:rw:./export/root,id=fat32,format=raw,if=none
```

should it be enough to change these `h` and `e` options in order to have this emulated device available on my LAN? I guess that I'm missing something.

razr, 2020-09-24:
I create a tap0 interface before I run QEMU
```
sudo apt-get install uml-utilities
sudo tunctl -u $USER -t tap0
sudo ifconfig tap0 192.168.200.254 up
```
then you can telnet to the QEMU target
```
telnet 192.168.200.1
```
or for the client
```
./examples/PublishHelloWorld/PublishHelloWorldClient 192.168.200.1 8888
```

I'm using 20.04 Ubuntu as a host machine

pablogs9, 2020-09-25:
Hello @razr, I have achieved the QEMU setup with the network, I also have successfully connected to the emulated using `ẁrdbg` following the instructions from [this link](https://labs.windriver.com/downloads/wrsdk-vxworks7-docs/VxWorksSDK-ApplicationDeveloperGuide.html#run-and-debug-from-cli):

```bash
(wrdbg) target connect vxworks7:TCP:192.168.200.1:1534 -kernel /wrsdk/bsps/itl_generic_2_0_2_1/boot/vxWorks 
Connecting...
Connected to TCP:192.168.200.1:1534
(wrdbg) file /work/export/root/llvm/bin/micro_ros_agent.vxe
Reading symbols from /work/export/root/llvm/bin/micro_ros_agent.vxe...done
[New Thread 1]
[New inferior 2 (0xffff80000063aae0)]
Switching to inferior 2 [0xffff80000063aae0] (micro_ros_agent.vxe)
```

But the [WRDBG Reference Guide](https://labs.windriver.com/downloads/toolkit/wrdbg_tools/lib/python/doc/html/wrdbg/wrdbg.html) referred in this page is down. Is possible to provide a comprehensive guide for using `wrdbg`?

Thank you very much.

razr, 2020-09-25:
@pablogs9, Thanks for looking into it.
it is rather in the SDK directory under $WIND_TOOLS/lib/python/doc/html/wrdbg/wrdbg.html

Meanwhile, I have merged your changes and applied missing patches.

razr, 2020-10-07:
@pablogs9, I have enabled logging, now it prints something like
```bash
[vxWorks *]# rtp exec -u 0x20000 micro_ros_agent.vxe "udp4 --port 8888 -v6"
Launching process 'micro_ros_agent.vxe' ...
Process 'micro_ros_agent.vxe' (process Id = 0xffff8000005bc1d0) launched.
Press CTRL+C to exit
[1602106537.450000] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1602106537.483333] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1602106634.216666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01
[1602106634.233333] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01
[1602106634.300000] info     | Root.cpp           | create_client            | create                 | client_key: 0xAAAABBBB, session_id: 0x81
[1602106634.316666] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2863315899, address: 192.168.200.254:45190
[1602106634.316666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01
[1602106634.333333] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAAAABBBB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1602106634.350000] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2863315899, address: 192.168.200.254:45190
[1602106634.350000] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 388, data: 
0000: 81 80 00 00 01 05 68 00 00 0A 00 11 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 64 65 66 61 75 6C 74 5F 78 72 63 65 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 01 05 64 00 00 0B 00 12 02 02 00 00 56 00 00 00
0080: 3C 64 64 73 3E 3C 74 6F 70 69 63 3E 3C 6E 61 6D 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 54 6F 70 69
00A0: 63 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 3C 2F 64 61
00C0: 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 11 01 05 0F 00 00 0C 00 13
00E0: 03 02 00 00 01 00 00 00 00 00 11 00 01 05 92 00 00 0D 00 15 05 02 00 00 84 00 00 00 3C 64 64 73
0100: 3E 3C 64 61 74 61 5F 77 72 69 74 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45
0120: 59 3C 2F 6B 69 6E 64 3E 3C 6E 61 6D 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 54 6F 70 69 63 3C 2F 6E
0140: 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 3C 2F 64 61 74 61 54 79
0160: 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E 3C 2F 64 64 73 3E 00
0180: 00 13 00 00
[1602106634.383333] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAAAABBBB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1602106634.383333] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2863315899, address: 192.168.200.254:45190
[1602106634.383333] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.400000] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAAAABBBB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1602106634.500000] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.516666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.516666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.516666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.516666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.533333] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1602106634.616666] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[... cut ...]
[1602106634.950000] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAAAABBBB, publisher_id: 0x001(3), participant_id: 0x001(1)
[1602106634.966666] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAAAABBBB, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 12 00 00
[1602106635.033333] debug    | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xAAAABBBB, datawriter_id: 0x001(5), publisher_id: 0x001(3)
```

razr, 2020-10-07:
and on the other side:
```
$ ./examples/PublishHelloWorld/PublishHelloWorldClient 192.168.200.1 8888 -v6
==>  24: (key: AA AA BB BB | n: 0 |  0) [CREATE CLIENT | session: 0x81 | key: AA AA BB BB ] t: 0ms 
==>  24: (key: AA AA BB BB | n: 0 |  0) [CREATE CLIENT | session: 0x81 | key: AA AA BB BB ] t: 0ms 
==>  24: (key: AA AA BB BB | n: 0 |  0) [CREATE CLIENT | session: 0x81 | key: AA AA BB BB ] t: 0ms 
<==  19: (key: AA AA BB BB | n: 0 |  0) [STATUS AGENT] t: 45ms 
==> 388: (key: AA AA BB BB | r:80 |  0) [CREATE | REPLACE | req: 0x000A | obj: 0x0011 | PARTICIPANT | xml: 89]
                                        [CREATE | REPLACE | req: 0x000B | obj: 0x0012 | TOPIC | obj: 0x0011 | xml: 86]
                                        [CREATE | REPLACE | req: 0x000C | obj: 0x0013 | PUBLISHER | obj: 0x0011 | xml: 1]
                                        [CREATE | REPLACE | req: 0x000D | obj: 0x0015 | DATAWRITER | obj: 0x0013 | xml: 132] t: 45ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 46ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 48ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 52ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 60ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 76ms 
<==  19: (key: AA AA BB BB | n: 0 |  0) [STATUS AGENT] t: 87ms 
<==  19: (key: AA AA BB BB | n: 0 |  0) [STATUS AGENT] t: 105ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 108ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 172ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 300ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 557ms 
<==  14: (key: AA AA BB BB | r:80 |  0) [STATUS | req: 0x000A | obj: 0x0011 | OK] t: 619ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [ACKNACK | stream: 0x80 | seq_num: 1 | bitmap: 0000000000000000] t: 619ms 
<==  14: (key: AA AA BB BB | r:80 |  1) [STATUS | req: 0x000B | obj: 0x0012 | OK] t: 647ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [ACKNACK | stream: 0x80 | seq_num: 2 | bitmap: 0000000000000000] t: 647ms 
==>  13: (key: AA AA BB BB | n: 0 |  0) [HEARTBEAT | stream: 0x80 | first: 0 | last: 0] t: 1069ms 
Error at create entities: participant: 0 topic: 0 publisher: 255 darawriter: 255
```

pablogs9, 2020-10-08:
Hello @razr, two questions:

- Is the logging enabled now by default following the instructions [here](https://docs.google.com/document/d/1G1hR4TwZkWrnwQO8E3KZSITtovbgPHlFiUXTxG8NmQc/edit) or I need to do something else?
- This Agent and client output is the one that I expected as the session is created correctly, the DDS participant and publisher are also created but a sort of segfault appears when creating the data writer. Do you also have this error? Can you provide me some information about what part of the code is producing this faulty memory access?

Thanks a lot

razr, 2020-10-08:
Hello @pablogs9,
* yes logging is enabled by default https://github.com/razr/Micro-XRCE-DDS-Agent/commit/1b076d5a184d5c4650a2de6ca92b9909df86dc8b. you can do
```
$ cd $HOME/github/Wind-River/vxworks7-ros2-build/build/ros2/ros2_ws/src/eProsima/Micro-XRCE-DDS-Agent
$ git pull
$ cd $HOME/github/Wind-River/vxworks7-ros2-build/
$ rm build/.stamp/micro-ros.install
$ rm build/.stamp/micro-ros.build
$ rm -rf build/ros2/ros2_ws/build/microxrcedds_agent/
$ make
```
if you build inside the docker container, you might need to add before (CLI11 complains if it is not set)
```
git config --global user.email "you@example.com"
git config --global user.name "Your Name"
```

* I do not see any segfault, the whole qemu VxWorks instance gets a reboot. I'd expect that the app will crash and create a coredump. I'll look into it later on today

pablogs9, 2020-10-08:
> I do not see any segfault, the whole qemu VxWorks instance gets a reboot. I'd expect that the app will crash and create a coredump. I'll look into it later on today

That is true also for me, in QEMU it reboots and in RPi its gives the `0xffff8000010bcbe0 (t40): RTP 0xffff800000d66560 has been deleted due to signal 11.`

yuzu-ogura, 2024-04-09:
> @pablogs9 thanks for providing patches. There are other two patches I need to apply and after that, I can build the whole project. It runs on my side without a crash as
>
> ``` [...]

hello @razr , I also occurred signal 11 error, when I execute other RTP programs.(my program), I'm eager to know how to trace the error explicitly, but the link [https://labs.windriver.com/downloads/wrsdk-vxworks7-docs/VxWorksSDK-ApplicationDeveloperGuide.html#run-and-debug-from-cli](url) is invalid.

yuzu-ogura, 2024-05-08:
> @yuzu-ogura sorry for the inconvenience, WRLabs site was done a while, now it is back again. could you please try it again.

Excuse me, I didn't find the guidence. I launch a rtp in QEMU(x86) and got signal 11. If wrgdb is a buildin command of VxWorks ?

razr, 2024-05-08:
`[vxWorks *]# rtp exec -u 100000 xmltest`

yuzu-ogura, 2024-05-08:
> `[vxWorks *]# rtp exec -u 100000 xmltest`

Thank you very much! You are so powerful! Is there any shortcuts or learning guides available for getting started with VxWorks?

yuzu-ogura, 2024-05-08:
Excuse me, I'm so clumsy. I met an error, could you tell me how to solve it?

razr, 2024-05-08:
is it a symlink? I always dereference symlinks with `-L` while copying them

```
$ sudo cp -r -L ./output/export/deploy/* ~/tmp/mount/.
```

yuzu-ogura, 2024-05-09:
> is it a symlink? I always dereference symlinks with `-L` while copying them
> 
> ```
> $ sudo cp -r -L ./output/export/deploy/* ~/tmp/mount/.
> ```

Thanks @razr .
Yes, it is a symlink. I followed your solution but not solved. I found the library file size is unbelievable, it up to 104MB. Maybe the file size too huge to be found? 
[image]

razr, 2024-05-09:
`libfastrtps.so.2.6` shouldn't be a symlink. Is it still? What error do you have now?

yuzu-ogura, 2024-05-09:
> `libfastrtps.so.2.6` shouldn't be a symlink. Is it still? What error do you have now?

Thanks @razr  .The error is still not found library, like below:
[image]
I have already dereference symlink:
[image]

razr, 2024-05-09:
please use `;` as a separator in the `LD_LIBRARY_PATH` instead of `:`, e.g. 

```bash
set env LD_LIBRARY_PATH="path1;path2"
```

yuzu-ogura, 2024-05-10:
> please use `;` as a separator in the `LD_LIBRARY_PATH` instead of `:`, e.g.
> 
> ```shell
> set env LD_LIBRARY_PATH="path1;path2"
> ```

Thank you so much @razr , I followed your advice, it turns out that overcome this problem.

razr, 2024-05-10:
glad to see it. Now back to your question about a library size. It is compiled in `Debug` mode. If you want to have it smaller please change CMAKE_BUILD_TYPE from `Debug` to `Release` here https://github.com/Wind-River/vxworks7-ros2-build/blob/master/pkg/ros2/files/humble/vxworks.mixin#L8

yuzu-ogura, 2024-05-11:
> I'm glad to hear it. Now back to your question about the library size. It is compiled in `Debug` mode. If you want to have it smaller please change CMAKE_BUILD_TYPE from `Debug` to `Release` here https://github.com/Wind-River/vxworks7-ros2-build/blob/master/pkg/ros2/files/humble/vxworks.mixin#L8

Thanks a lot @razr , I get it.
Now I have one more question. This is my QEMU command :
`qemu-system-x86_64 -m 2G -kernel vxsdk/bsps/itl_generic_3_0_0_2/vxWorks -net nic -net user,hostfwd=tcp::1534-:1534,hostfwd=tcp::2345-:2345 -display none -serial stdio -monitor none -append "bootline:fs(0,0)host:vxWorks h=10.0.2.2 e=10.0.2.15 u=target pw=vxTarget o=gei0"`
How to open multiple terminals under VxWorks like Linux. 
(we can see here are double terminals)
[image]

razr, 2024-05-11:
I'm not so sure about your setup. I do `telnet` to the target as described https://github.com/Wind-River/vxworks7-ros2-build?tab=readme-ov-file#telnet-to-the-vxworks-qemu-target