Wind-River/vxworks7-ros2-build #10: Release date ROS2 foxy support / Support information

DentOpt, 2021-01-20:
Interesting work!
Regarding the VXworks ROS2 support
- is already a release date for the ros2-foxy support?
- which DDS vendors are supported?
- opencv supported 
- limitations of VXworks ROS2 bridge?

Thanks

razr, 2021-01-20:
Thanks @DentOpt, to answer your questions 

- I have started porting on foxy a while back, see https://github.com/Wind-River/vxworks7-ros2-build/tree/foxy-SR0640, do you have a particular need in foxy? I was thinking to align to galactic instead.
- we currently support FastRTPS, and CycloneDDS comes as the next one.
- OpenCV for VxWorks is supported, see https://labs.windriver.com/opencv/, 
- not sure I understand your question, there is no bridge as such, ROS2 middleware is ported to the VxWorks

DentOpt, 2021-01-20:
Thank you for this clarification @Razr, that helps. 
Does this mean that the foxy support is stable?
For us, foxy is interesting due to it's long support until 2023.

razr, 2022-09-26:
superseded by `humble`, see #47

closed 2022-09-26 by razr