Wind-River/vxworks7-ros2-build #11: Build failure for hls_lfcd_lds_driver package

sdoerr20, 2021-02-18:
According to a rebuild of the ROS2 packages in my TurtleBot3 project lead to
an unexpected error in the hls_lfcd_lds_driver package while no changes to the
existing toolchain have been made.

After downloading the WindRiver SDK für the RPI3 in version **1.5** and
checking out the ROS2 branch **dashing-release-SR0640-python**, I was able
to reproduce the failing build. The build trace showed the following error.

~~~
Starting >>> hls_lfcd_lds_driver
Finished <<< message_filters [5.05s]
Starting >>> tf2_ros
--- stderr: hls_lfcd_lds_driver
/work/build/ros2/ros2_ws/src/utils/hls_lfcd_lds_driver/src/hlds_laser_publisher.cpp:189:3: error: expected unqualified-id
  catch (boost::system::system_error& ex)
  ^
/work/build/ros2/ros2_ws/src/utils/hls_lfcd_lds_driver/src/hlds_laser_publisher.cpp:194:1: error: extraneous closing brace ('}')
}
^
In file included from /work/build/ros2/ros2_ws/src/utils/hls_lfcd_lds_driver/src/hlds_laser_publisher.cpp:37:
/work/build/ros2/ros2_ws/src/utils/hls_lfcd_lds_driver/include/hls_lfcd_lds_driver/lfcd_laser.hpp:72:11: warning: private field 'motor_speed_' is not used [-Wunused-private-field]
        uint16_t motor_speed_; ///< @brief current motor speed as reported by the LFCD.
                 ^
1 warning and 2 errors generated.
make[4]: *** [CMakeFiles/hlds_laser_publisher.dir/src/hlds_laser_publisher.cpp.obj] Error 1
make[3]: *** [CMakeFiles/hlds_laser_publisher.dir/all] Error 2
make[2]: *** [all] Error 2
---
Failed   <<< hls_lfcd_lds_driver [5.88s, exited with code 2]
Aborted  <<< tf2_ros [3.97s]

Summary: 75 packages finished [2min 31s]
  1 package failed: hls_lfcd_lds_driver
  1 package aborted: tf2_ros
  9 packages had stderr output: dynamixel_sdk geometry_msgs hls_lfcd_lds_driver nav_msgs rmw_cyclonedds_cpp sensor_msgs std_srvs tf2_msgs turtlebot3_msgs
  1 package not processed
Makefile:84: recipe for target 'turtlebot3.build' failed
make[1]: *** [turtlebot3.build] Error 2
make[1]: Leaving directory '/work/pkg/turtlebot3'
make: *** [Makefile:19: all] Error 1
~~~

Looking at code of **hlds_laser_publisher.cpp** (see below) the formatting of
the try-catch block in the main-method is not correct. As temporary fix I just
used comments to solve the braces-issue. It would be helpful, to solve this
misspelling in a further git patch, so the build does not require two tries.

~~~
int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);
  fd = open("/usb2ttyS/1", O_RDWR);
  if (fd < 0) {
        printf("Cannot open %s\n","/usb2ttyS/1");
  }
  auto node = rclcpp::Node::make_shared("hlds_laser_publisher");
  rclcpp::Publisher<sensor_msgs::msg::LaserScan>::SharedPtr laser_pub;
  std::string port;
  std::string frame_id;
  int baud_rate;
  node->declare_parameter("port");
  node->declare_parameter("frame_id");

  node->get_parameter_or<std::string>("port", port, "/usb2ttyS/1");
  node->get_parameter_or<std::string>("frame_id", frame_id, "laser");
  baud_rate = 230400;
  RCLCPP_INFO(node->get_logger(), "Init hlds_laser_publisher Node Main");
  RCLCPP_INFO(node->get_logger(), "port : %s frame_id : %s", port.c_str(), frame_id.c_str());
// MISSING try{ expression
// try {
    hls_lfcd_lds::LFCDLaser laser(port, baud_rate);
    laser_pub = node->create_publisher<sensor_msgs::msg::LaserScan>("scan", rclcpp::QoS(rclcpp::SensorDataQoS()));
    auto scan = std::make_shared<sensor_msgs::msg::LaserScan>();
    while (rclcpp::ok())
    {
      scan->header.frame_id = frame_id;
      laser.poll(scan);
      scan->header.stamp = node->now();
      laser_pub->publish(*scan);
    }
    laser.close();

    return 0;
//  }
// unknown exception to catch
//  catch (boost::system::system_error& ex)
//  {
    //ROS_ERROR("An exception was thrown: %s", ex.what());
//    return -1;
//  }
}
~~~

razr, 2021-02-22:
hi @sdoerr20, thanks for reporting it. 
We do support the SR0640 VxWorks release with ROS2. Please use VxWorks SDK for Raspberry Pi 3 version 1.4 - https://labs.windriver.com/downloads/wrsdk-vxworks7-raspberrypi3b-1.4.tar.bz2 
Is it any particular reason why you have selected the SR0650 SDK?

sdoerr20, 2021-02-24:
Hi @razr, I am also took a look at the RPI3 SDK in version, as this is the first
version mentioning a VS code plug-in. We are currently thinking of offering
VS code as second IDE for RTP development.

For completeness I have also tested version **1.4** now with the same result.

razr, 2021-02-28:
thanks, @sdoerr20 for reporting it. Indeed one patch fails to apply. it is fixed now https://github.com/Wind-River/vxworks7-layer-for-ros2/commit/0c9c6905ec3d3a6365056777cceba54bc58a071e

sdoerr20, 2021-03-01:
@razr Thanks for the update here. I was able to run the ROS2 build without stopping because of an error message.

closed 2021-03-01 by razr

GitHubBoyz, 2022-09-06:
I create a turtlebot3 in ros2galactic it's almost done but one small problem is occurred in colcon build --symlink -install 
Error:1 package failed:hls_lfcd_lds_driver 
Please any one give a solution for this problem