Wind-River/vxworks7-ros2-build #13: Reference of git dashing-devel branch in turtlebot3 + util repos

Prof-Florian-Kuenzner, 2021-06-07:
Hello,

Yesterday, the build of the ros2-build `dashing-release-SR0640-python` run into some build errors, which was caused by too new versions of turtlebot3 + utils (DynamixelSDK +hls_lfcd_lds_driver), where the patch files didn't work correctly.

After changing the `pkg/turtlebot3/Makefile` line 36 to `PKG_REPO_COMMIT_ID = dashing-devel` everything works fine.

Is there a reason, why `PKG_REPO_COMMIT_ID = ros2-devel` is used here: https://github.com/Wind-River/vxworks7-ros2-build/blob/882173a2cc9377decec42397cb669760304c8cfb/pkg/turtlebot3/Makefile#L36

razr, 2021-06-08:
@Prof-Florian-Kuenzner thanks for reporting it. In the past, there was only ROS2 branch ```ros2-devel```. I'll fix it this week.

Prof-Florian-Kuenzner, 2021-08-31:
@razr thank you. This looks good. I think we can close the issue now.

commit 9a1ece23 (2022-06-28): Merge pull request #15 from Wind-River/dashing-release-SR0640-python

fix #13
commit afd1178f (2022-06-28): fix #13
closed 2021-08-31 by Prof-Florian-Kuenzner