Wind-River/vxworks7-ros2-build #28: ros2 native tools is not working

enriLoniterp, 2022-07-13:
Hi @razr, hope you're doing fine.

I've seen your changes to the README and i followed the native ROS 2 compilation instructions but it fails.
I don't know if you remember my current situation but i'm still not able to launch my packages and my nodes inside vxworks platform using qemu. i have my project locally which i run correctly but if i attemp to run it i don't know the exact procedure. i had tried different solutions like move the artifact build in the install folder to the /export/bin dir and then use rtp exec -u 0x20000 ros 2 python3 run cpp_pubsub talker but it didn't work.

Now i think that the solution is to build my package with the new instructions you made and then run the package in the same way before. Am i correct? because if use `ros2 pkg list` in the /ata4/lib dir i don't see any package so maybe i build it with this instructions you wrote and then i run on vxworks.

I have this error on my vm:

```
wruser@3d2363ccb1f4:/work/build/ros2/ros2_native$ colcon build --merge-install --cmake-force-configure --packages-up-to-regex examples_rcl* ros2action ros2component ros2msg ros2node ros2pkg ros2service ros2topic ros2cli ros2lifecycle ros2multicast ros2param ros2run ros2srv pendulum_control --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_TESTING:BOOL=OFF

##a lot of packages building with some warnings
```
Starting >>> poco_vendor
--- stderr: poco_vendor                                               
CMake Error at /wrsdk/toolkit/host_tools/x86_64-linux/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PCRE (missing: PCRE_LIBRARY PCRE_INCLUDE_DIR)
Call Stack (most recent call first):
  /wrsdk/toolkit/host_tools/x86_64-linux/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindPCRE.cmake:23 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  Foundation/CMakeLists.txt:35 (find_package)

make[2]: *** [poco-1.8.0.1-release-prefix/src/poco-1.8.0.1-release-stamp/poco-1.8.0.1-release-configure] Error 1
make[1]: *** [CMakeFiles/poco-1.8.0.1-release.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< poco_vendor [3.38s, exited with code 2]
Aborted  <<< fastrtps [2min 27s]                                            

Summary: 42 packages finished [2min 50s]
  1 package failed: poco_vendor
  1 package aborted: fastrtps
  14 packages had stderr output: ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_package ament_pep257 ament_uncrustify ament_xmllint cyclonedds opensplice_cmake_module poco_vendor
  106 packages not processed
```
```
I am missing something of the procedure of How to build and run my package on VxWorks?
Thanks for your attention!

enriLoniterp, 2022-07-13:
hi @razr,

re-run the ros2 native situation and got this different issues:

```
wruser@cd9bc46bb25b:/work/build/ros2/ros2_native$ colcon build --merge-install --cmake-force-configure --packages-up-to-regex examples_rcl* ros2action ros2component ros2msg ros2node ros2pkg ros2service ros2topic ros2cli ros2lifecycle ros2multicast ros2param ros2run ros2srv pendulum_control --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_TESTING:BOOL=OFF
Starting >>> ament_package
Starting >>> ament_lint                       
Starting >>> ament_cppcheck
Starting >>> osrf_testing_tools_cpp
--- stderr: ament_lint                                                                                                                      
/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< ament_lint [1.86s]
Starting >>> ament_flake8
--- stderr: ament_package                                                                                                                     
/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< ament_package [2.02s]
Starting >>> ament_cmake_core
--- stderr: ament_cppcheck                                                                                                                       
/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< ament_cppcheck [2.20s]
Starting >>> fastcdr
Finished <<< ament_cmake_core [1.68s]                                                                                                                       
Starting >>> ament_cmake_libraries
--- stderr: ament_flake8
/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< ament_flake8 [1.76s]
Starting >>> ament_pep257
--- stderr: osrf_testing_tools_cpp                                                                                                                             
In file included from /work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/test_runner/main.cpp:23:
/work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/test_runner/./execute_process.hpp:101:15: error: use of undeclared identifier 'fork'
  pid_t pid = fork();
              ^
/work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/test_runner/./execute_process.hpp:116:15: error: use of undeclared identifier 'execvp'
    int ret = execvp(cmd, arguments.data());
              ^
2 errors generated.
make[2]: *** [src/test_runner/CMakeFiles/test_runner.dir/main.cpp.o] Error 1
make[1]: *** [src/test_runner/CMakeFiles/test_runner.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/custom_memory_functions.cpp:26:
/work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/./count_function_occurrences_in_backtrace.hpp:30:10: fatal error: 'execinfo.h' file not found
#include <execinfo.h>
         ^~~~~~~~~~~~
1 error generated.
make[2]: *** [src/memory_tools/CMakeFiles/memory_tools.dir/custom_memory_functions.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/memory_tools_service.cpp:20:
In file included from /work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/./stack_trace_impl.hpp:31:
/work/build/ros2/ros2_native/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/./vendor/bombela/backward-cpp/backward.hpp:749:24: error: use of undeclared identifier 'backtrace'
    size_t trace_cnt = backtrace(&_stacktrace[0], _stacktrace.size());
                       ^
1 error generated.
make[2]: *** [src/memory_tools/CMakeFiles/memory_tools.dir/memory_tools_service.cpp.o] Error 1
make[1]: *** [src/memory_tools/CMakeFiles/memory_tools.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< osrf_testing_tools_cpp [3.99s, exited with code 2]
Aborted  <<< ament_pep257 [0.73s]                                                                                               
Aborted  <<< ament_cmake_libraries [0.86s]                                                                
Aborted  <<< fastcdr [2.42s]                             

Summary: 5 packages finished [6.40s]
  1 package failed: osrf_testing_tools_cpp
  3 packages aborted: ament_cmake_libraries ament_pep257 fastcdr
  6 packages had stderr output: ament_cppcheck ament_flake8 ament_lint ament_package fastcdr osrf_testing_tools_cpp
  141 packages not processed
```

razr, 2022-07-13:
@enriLoniterp, the native ROS2 build was added as a reference to compare ROS2 behavior under Ubuntu and VxWorks. You need to rebuild the docker image for that since it was modified, see https://github.com/Wind-River/vxworks7-ros2-build/commit/76603d623b98a12e0256b3943cb4d8bc9060c53c

```bash
docker build --no-cache -t vxros2build:1.0 .
```

enriLoniterp, 2022-07-14:
Hi @razr,
ROS2 NATIVE COMPILATION problem **SOLVED**:
rebuild as you mentioned and bypassed sourcing env. operation and worked well, the example you provided worked and it is printed HelloWord! messages.

But still i can't understand what is the procedure to construct something mine with ROS2 inside VxWorks.
I can run each example you provided, but now i want to develop a little project by myself. My intentions is to build an arm-robotic application on VxWorks. Theoretically i should need my ROS 2 workspace, my packages and each package should contain nodes that i can compile and then launch all nodes to develop something useful IN VxWorks, so running on [VxWorks] terminal. Currently i don't understand the structure of the entire system i've built and the development workflow.

I know that `/export/deploy` has bins like `timer lambda` compiled for VxWorks. So my idea was "i built my architecture in local with colcon, move all my executable to export/deploy and run it". For example i tried moving MYTalker inside /export/deploy but VxWorks can't execute it. It is not the solution i guess.

Last i've noticed that inside build there is a directory called /ros2_ws (/home/rosdev/VxWorks_ros2_master/vxworks7-ros2-build/build/ros2/ros2_ws) with a workspace, maybe i should place my package and do twice all the procedure from scratch.

I am really sorry that i'm asking you all this questions but i'm stucked at this level and i want to go forward and develop my own architecture.

Thank you again, hope it's ok if ask you some questions.

Enrico

razr, 2022-07-14:
`python3` is located in the `/ata4/bin` directory which is a non-standard path and you need whether  `cd /ata4/bin` to it or run it with a full path `/ata4/bin/python3`. For the same reason, you also need to `set env LD_LIBRARY_PATH=/ata4/lib"`

```bash
-> cmd
[vxWorks *]# cd /ata4/bin
[vxWorks *]# set env LD_LIBRARY_PATH=/ata4/lib"
[vxWorks *]# set env AMENT_PREFIX_PATH="/ata4"
[vxWorks *]# rtp exec -u 0x20000 python3 /ata4/lib/demo_nodes_py/talker
```

enriLoniterp, 2022-07-20:
hi @razr,
sorry if i am late.
As i understood from your explantation, i've taken my exe from my local machine the directory containing my publisher and subscriber and placed inside `/export/deploy/lib`. Done all the process from scratch and launched qemu with VxWorks, results:
```
-> cmd
[vxWorks *]# cd /ata4/bin
[vxWorks *]#    set env LD_LIBRARY_PATH="/ata4/lib"
[vxWorks *]#    set env AMENT_PREFIX_PATH="/ata4"
[vxWorks *]# rtp exec -u 0x20000 python3 /ata4/lib/cpp_pubsub/talker
Launching process 'python3' ...
Process 'python3' (process Id = 0xffff8000082692a0) launched.
  File "/ata4/lib/cpp_pubsub/talker", line 1
SyntaxError: Non-UTF-8 code starting with '\xe5' in file /ata4/lib/cpp_pubsub/talker on line 2, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
[vxWorks *]# rtp exec -u 0x20000 python3 /ata4/lin/?
Launching process 'python3' ...
Process 'python3' (process Id = 0xffff8000083c8040) launched.
./python3: can't open file '/ata4/lin/': [Errno 2] no such file or directory
[vxWorks *]# rtp exec -u 0x20000 python3 /ata4/lib/demo_nodes_py/talker
Launching process 'python3' ...
Process 'python3' (process Id = 0xffff8000083c8040) launched.
./python3: can't open file '/ata4/lib/demo_nodes_py/talker': [Errno 2] no such file or directory
[vxWorks *]# 
```
I think something wrong, i desire bring my packages on my simulation, inside VxWorks and run it, i can't understand how.
Thank you for your attention, sorry for disturbing!

razr, 2022-07-22:
Sorry for the confusion, I thought you try to run python scripts.
if your nodes are written in C++ and are not the python scripts you need to put them under `/ata4/bin` directory and run

```bash
[vxWorks *]# cd /ata4/bin
[vxWorks *]# set env LD_LIBRARY_PATH="/ata4/lib"
[vxWorks *]# rtp exec -u 0x20000 /ata4/bin/talker
```

enriLoniterp, 2022-07-28:
hi @razr sorry if i'm late,
with 'your nodes' you mean my piece of C++ code or my executable talker got from previous compilation in my local ws? and if it's the second which should i compile my nodes, locally or on VxWorks platform?

Thank you very much for your patient and kindness, 
Enrico

enriLoniterp, 2022-07-28:
I have another question, i'm wondering which ROS 2 version should i use to develop my own application? To not have compatibility issues in my simulation

razr, 2022-07-29:
fixed in #26

closed 2022-07-29 by razr