Wind-River/vxworks7-ros2-build #39: [vxWorks *]# python3 ros2 launch not working

enriLoniterp, 2022-08-04:
Hi @razr ,
I highly thank you for the massive work you've made since now my node run on VxWorks and i'm excited to continue this experience!

I'm now ahead the one process and one node, i desire two nodes communicating. My idea was to use `ros2 launch my_package my_script.launch.py ` **but**:
- on nativeROS2 it wasn't finding ros2 launch, i've just add in 
`wruser@ros2native:/work/ros2_native$ colcon build --merge-install --cmake-force-configure --packages-up-to-regex examples_rcl* ros2action ros2component ros2msg ros2node ros2pkg ros2service ros2topic ros2cli **ros2launch** ros2lifecycle ros2multicast ros2param ros2run ros2srv pendulum_control --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_TESTING:BOOL=OFF` and it let me launched in the same bash, in the same QEMU simulation my two nodes

-Then i went for the virtual ambient on qemu and neverthless it worked on the ROS2 native it wan't finding my command. So after a few search i went to ../vxworks7-ros2-build/pkg/ros2/packages.mk and modified this line. 
```

 ROS2_PYTHON_TOOLS=ros2action ros2component ros2msg ros2node ros2pkg \
	ros2service ros2launch ros2topic ros2cli ros2lifecycle ros2multicast ros2param \
	ros2run ros2srv

```
After another rebuilding `ros2 launch `worked, maybe it is only this two small error or it could be missing something(?)

But i have few questions, it is possible to have a solution without ros2 launch but with many qemu simulation actived and communicating?

Other thing important, will you release an updated version for **ROS2 Foxy**? i was intended to combine ROS2 on VxWorks with EtherCATMaster, an open source project which simulate an EtherCAT Master in order to control a 6 axis robot with ROS2 Control(maybe also integrating MoveIt2), the thing is that EthercatMaster-ROS2 Control integration has been developed at least at ROS2 Foxy.

Thank you very much again!

razr, 2022-08-04:
@enriLoniterp  I'm glad to see that everything is working now. `ros2 launch` is on my list, however not the first prio. You can provide a PR if you like.

razr, 2023-12-04:
fixed in #100

closed 2023-12-04 by razr