Wind-River/vxworks7-ros2-build #43: VXWorks7 update on ROS2 Humble?

enriLoniterp, 2022-08-30:
Hi @razr,
i hope you enjoy the summer and you are fine.

I'm working on my personal project and i would like to know if you have plans for porting ROS2 Humble to VXworks7. My architecture is based on Ros2Humble, i have available a MoveIt! project for a pick and place application with my "personal" robot, also this robot should communicate with the Servo motor through EtherCAT connection and luckily i have one EtherCAT master working with ROS 2 Control. All this packages and library should work alongside Ros2 Humble.

So, i ask you if you have in mind to add ros2 humble and if it is, should i add my packages as described in **VxWorks ROS2 development** or for MoveIt and ROS2 Control packages i should modify you code directly and build VxWorks directly with them(Moveit and ros2Control package).

Thank you again for help, 
hope everything ok

Enrico

razr, 2022-09-04:
I have a first working prototype [here](https://github.com/razr/vxworks7-ros2-build/tree/humble-release-22.03). I have never tried to port `Movit`. I'd recommend starting with a native ROS2 development to bring everything together.

enriLoniterp, 2022-09-05:
Hi @razr ,

I'm going to test this prototype as soon as possible, thank you very much for the work.

I agree with you about the native ROS2 development, but it would be amazing to port `MoveIt`.
If you don't mind, could you explain me how to port it?

Enrico

enriLoniterp, 2022-09-15:
Hi @razr,

i'm just trying to replicate what achieved in the version for dashing diademata on Humble version downloading your protype bur fiast problem occured:

on` sudo docker build -t vxros2build:humble Docker/vxros2build/.`
```

[sudo] password for rosdev: 
Sending build context to Docker daemon  10.24kB
Step 1/16 : FROM vxbuild:22.04
pull access denied for vxbuild, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

I don't is anything about me but i can proceed looking for it online.

Thanks you again for the help, 
Enrico

razr, 2022-09-19:
@enriLoniterp, have you built `vxbuild:22.04` before?
```bash
docker build -t vxbuild:22.04 Docker/vxbuild/.
docker build -t vxros2build:humble Docker/vxros2build/.
```

enriLoniterp, 2022-09-19:
Hi @razr,

previous problem solved! (my bad on my pc)

Now i'm struggling with sourcing environment:
```

 sudo docker run -ti -v /home/rosdev/Desktop/wrsdk-vxworks7-up2:/wrsdk -v /home/rosdev/VxWorks_ros2_master/vxworks7-ros2-build:/work vxros2build:humble
wruser@0a68dae15d53:/work$ ls /wrsdk
NCLA.txt  README.md  bsps  docs  examples  toolkit
wruser@0a68dae15d53:/work$ ls /wrsdk/toolkit
bin  compilers  host_tools  include  lib  license  sdk_tools  wind_sdk_env.linux  wrdbg_tools
wruser@0a68dae15d53:/work$ ls /wrsdk/examples
Makefile  cmakefiles  cobble  hello_world  makefiles  morse  philosophers
wruser@0a68dae15d53:/work$ 
```
As you can see there is no file as you described in the README in this specific folder but there is on `wind_sdk_env.linux `

enriLoniterp, 2022-09-19:
Hi @razr ,
solved rev problem (my side problem)

Now i've installed everything new and follower humble README.

you made create a dir in Download called wrsdk but then in docker run you haven't written ( think adding it is correct)

Looked like this:
```
sudo docker run -ti -v ~/Downloads/wrsdk/wrsdk-vxworks7-qemu:/wrsdk -v $PWD:/work vxros2build:humble
wruser@80394c3793a1:/work$ ls /wrsdk
NCLA.txt  README.md  compilers  docs  examples  sdk.json  sdk.yml  sdkenv.sh  tools  vxsdk
wruser@80394c3793a1:/work$ source /wrsdk/sdkenv.sh
wruser@80394c3793a1:/work$ make
Makefile:20: *** WIND_CC_SYSROOT is not set, please source the environment.  Stop.
wruser@80394c3793a1:/work$ make^C
wruser@80394c3793a1:/work$ exit
```
i do remember this error was a deal also in master branch but can't remember what it's be done to resolve it.
Thanks a lot!

Enrico

razr, 2022-09-20:
@enriLoniterp, you should check out the `humble-release-22.03` branch from my forked repo. However, I'd recommend waiting until I merge it to the main repo.

enriLoniterp, 2022-09-21:
Hi @razr,
the piece of code i shared it is about` humble-release-22.03 branch`  (ROS2 Humble)
But I do remember dealing with the same problem for Dashing!

If i'm not disturbing, when did you plan the merging on the main repo?

Thanks a lot,
Enrico

razr, 2022-09-22:
@enriLoniterp, 

You have this error
```bash
wruser@80394c3793a1:/work$ make
Makefile:20: *** WIND_CC_SYSROOT is not set, please source the environment.  Stop.
```
but there is no such message on line 20
https://github.com/razr/vxworks7-ros2-build/blob/humble-release-22.03/Makefile#L20

I plan to merge it this week

enriLoniterp, 2022-09-26:
Hi @razr,
i've seen you merged humble branch to main!

I think i should close this issue and maybe try a new one eventually after i tested this version!

Thank you very very much again !

Enrico

closed 2023-08-11 by razr