Wind-River/vxworks7-ros2-build #45: Error:1 package failed:hls_lfcd_lds_driver

razr, 2022-09-06:
I create a turtlebot3 in ros2galactic it's almost done but one small problem is occurred in colcon build --symlink -install 
Error:1 package failed:hls_lfcd_lds_driver 
Please any one give a solution for this problem

_Originally posted by @GitHubBoyz in https://github.com/Wind-River/vxworks7-ros2-build/issues/11#issuecomment-1238445182_

razr, 2022-09-06:
@GitHubBoyz do you build it for VxWorks or natively?

GitHubBoyz, 2022-09-08:
Ubuntu 20.04

razr, 2022-09-08:
@GitHubBoyz I see. It is better to ask your question [here](https://github.com/GitHubBoyz). I do assume, you are using https://github.com/ROBOTIS-GIT/turtlebot3/tree/galactic-devel to build it. This project is about porting ROS2 to VxWorks.

closed 2022-09-19 by razr