Wind-River/vxworks7-ros2-build #48: CMake Error at /wrsdk/vxsdk/host/x86_64-linux/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):

enriLoniterp, 2022-09-28:
hi @razr,
just followed new tutorials in each step but:

```
--- stderr: fastrtps
CMake Warning (dev) at /wrsdk/vxsdk/host/x86_64-linux/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:44 (find_package_handle_standard_args)
  cmake/common/eprosima_libraries.cmake:96 (find_package)
  CMakeLists.txt:220 (eprosima_find_thirdparty)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /wrsdk/vxsdk/host/x86_64-linux/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /wrsdk/vxsdk/host/x86_64-linux/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindThreads.cmake:205 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/cpp/CMakeLists.txt:245 (find_package)

---
Failed   <<< fastrtps [0.36s, exited with code 1]

```

enriLoniterp, 2022-09-28:
Other thing:
1. vxworks7-ros2-build/Docker/**22.04**/ vxbuild, 22.04 is missing from the path you signed.
2. vxworks7-ros2-build/Docker/**22.04**/ vxros2build, 22.04 is missing from the path you signed.
3. miss sudo, think it would be better

Other thing:

1. it asks me:
```
can't find file to patch at input line 16
Perhaps you used the wrong -p or --strip option?
File to patch: 
Skip this patch? [y]
```

razr, 2022-09-28:
It should not build `fastrtps`. I assume your docker container maybe still `dashing` ?
```bash
wruser@28429adc8c02:/work$ echo $ROS_DISTRO
humble
```
there is a GitHub action `humble` build here: https://github.com/Wind-River/vxworks7-ros2-build/actions/runs/3123243027

Please provide a complete list of step you have used.

enriLoniterp, 2022-09-29:
You are right, it prints me `dashing` instead of `humble` but i've followed every step you've written and built new containers, not the old ones made of dashing. How is it possible?

enriLoniterp, 2022-09-29:
> It should not build `fastrtps`. I assume your docker container maybe still `dashing` ?
> 
> ```shell
> wruser@28429adc8c02:/work$ echo $ROS_DISTRO
> humble
> ```
> 
> there is a GitHub action `humble` build here: https://github.com/Wind-River/vxworks7-ros2-build/actions/runs/3123243027
> 
> Please provide a complete list of step you have used.

I'm in the `master` branch:

```
git clone https://github.com/Wind-River/vxworks7-ros2-build.git
cd vxworks7-ros2-build
```
```
docker build -t vxbuild:22.04 Docker/22.04/vxbuild/.
docker build -t vxros2build:humble Docker/22.04/vxros2build/.
```
You actually have also `18.04`
Then i download new SDK:
```
cd ~/Downloads 
wget https://d13321s3lxgewa.cloudfront.net/wrsdk-vxworks7-qemu-1.10.tar.bz2
mkdir ~/Downloads/wrsdk && cd ~/Downloads/wrsdk
```
Don't use tar –jxvf ~/Downloads/wrsdk-vxworks7-qemu-1.10.tar.bz2 --strip 1 because my bash can't find --strip 1 option but extracted manually with Archive Manager
```
cd vxworks7-ros2-build
docker run -ti -v ~/Downloads/wrsdk-vxworks7-qemu/wrsdk:/wrsdk -v $PWD:/work vxros2build:humble
```
but it prints me `dashing` as i stated in the previous comment!

razr, 2022-09-30:
As described in the README

```bash
docker build -t vxbuild:22.04 Docker/22.04/vxbuild/.
docker build -t vxros2build:humble Docker/22.04/vxros2build/.
docker run -ti vxros2build:humble
wruser@a1412dc0a60c:/work$ echo $ROS_DISTRO
humble
wruser@a1412dc0a60c:/work$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.1 LTS"
```

closed 2022-10-03 by razr