Wind-River/vxworks7-ros2-build #76: Make failed in docker container

zzhou292, 2023-08-02:
I think there is a ROS2 related issue caused by cyclonedds. Any help will be really appreciated ! Thank you!!

The following error messages were generated after I sourced sdkenv.sh and 'make' command. I am using ROS2 humble, host system is ubuntu 22.04:

Starting >>> ament_cmake_export_libraries                
--- stderr: cyclonedds                                   
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:252:36: error: expected ')'
    GVTRACE ("started new thread %"PRIdTID": %s\n", ddsrt_gettid (), thrst->name);
                                   ^
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:252:5: note: to match this '('
    GVTRACE ("started new thread %"PRIdTID": %s\n", ddsrt_gettid (), thrst->name);
    ^
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/security/api/../../core/ddsi/include/dds/ddsi/q_log.h:25:29: note: expanded from macro 'GVTRACE'
#define GVTRACE(...)        DDS_CTRACE (&gv->logconfig, __VA_ARGS__)
                            ^
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/include/dds/ddsrt/log.h:450:28: note: expanded from macro 'DDS_CTRACE'
  DDS_CLOG_MSVC_WORKAROUND(DDS_CLOG(DDS_LC_TRACE, __VA_ARGS__))
                           ^
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/include/dds/ddsrt/log.h:419:18: note: expanded from macro 'DDS_CLOG'
      dds_log_cfg((cfg), (cat), __FILE__, __LINE__, DDS_FUNCTION, __VA_ARGS__) : 0)
                 ^
/work/build/ros2/ros2_ws/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:288:55: warning: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
static struct thread_state *get_available_thread_slot ()
                                                      ^
                                                       void
1 warning and 1 error generated.
make[4]: *** [src/core/CMakeFiles/ddsc.dir/build.make:961: src/core/CMakeFiles/ddsc.dir/ddsi/src/q_thread.c.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [CMakeFiles/Makefile2:368: src/core/CMakeFiles/ddsc.dir/all] Error 2
make[2]: *** [Makefile:159: all] Error 2
---
Failed   <<< cyclonedds [1.43s, exited with code 2]
Aborted  <<< rpyutils [2.14s]
Aborted  <<< ament_cmake_export_libraries [0.13s]
Aborted  <<< osrf_pycommon [1.24s]

Summary: 9 packages finished [8.20s]
  1 package failed: cyclonedds
  3 packages aborted: ament_cmake_export_libraries osrf_pycommon rpyutils
  6 packages had stderr output: ament_index_python ament_package cyclonedds domain_coordinator rosidl_cli rpyutils
  135 packages not processed
make[1]: *** [Makefile:104: ros2.build] Error 2
make[1]: Leaving directory '/work/pkg/ros2'
make: *** [Makefile:22: all] Error 1

razr, 2023-08-08:
Thanks for reporting it.

I see that `foonathan_memory_vendor` fails on the latest build as well.

```bash
...
Summary: 35 packages finished [41.7s]
  1 package failed: foonathan_memory_vendor
  1 package aborted: cyclonedds
```

razr, 2023-08-11:
I cannot reproduce your error. `foonathan_mememory_vendor` error was fixed [here](https://github.com/Wind-River/vxworks7-layer-for-ros2/pull/12). Both builds have finished successfully, please see [GHA](https://github.com/Wind-River/vxworks7-ros2-build/actions)

mkrunic-windriver, 2023-09-07:
> 

I have the similar issue with `make` in docker on `humble-release-23.03`, host Ubuntu 22.04:

```
--------------------------------------------------------------------------------
Building
--------------------------------------------------------------------------------
cd /work/build/ros2/ros2_ws; \
export COLCON_HOME=/work/build/ros2/ros2_ws/.colcon; \
colcon build --mixin vxworks --packages-up-to-regex examples_rclcpp_minimal_timer examples_rclcpp_minimal_client examples_rclcpp_minimal_service examples_rclcpp_minimal_publisher examples_rclpy_* ros2action ros2cli ros2component ros2doctor ros2interface ros2lifecycle ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2topic
/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_devtools/verb/extension_points.py:4: UserWarning: 'colcon_core.entry_point' has been deprecated, use 'colcon_core.extension_point' instead
  from colcon_core.entry_point import EXTENSION_POINT_GROUP_NAME
Starting >>> ament_package
Starting >>> gtest_vendor                     
Starting >>> domain_coordinator                                                 
Starting >>> ament_index_python
Starting >>> rosidl_cli                                                                                                                 
Starting >>> fastcdr
Starting >>> rpyutils                                                                                           
Starting >>> cyclonedds
--- stderr: domain_coordinator   
```
.
.
.
```
SyntaxError: invalid syntax
---
Failed   <<< rosidl_cli [2.01s, exited with code 1]
Aborted  <<< ament_package [2.92s]                                                                                          
Aborted  <<< fastcdr [2.60s]
Aborted  <<< rpyutils [2.56s]
Aborted  <<< gtest_vendor [2.86s]
Aborted  <<< cyclonedds [2.51s]
                                  
Summary: 0 packages finished [9.81s]
  3 packages failed: ament_index_python domain_coordinator rosidl_cli
  5 packages aborted: ament_package cyclonedds fastcdr gtest_vendor rpyutils
  4 packages had stderr output: ament_index_python domain_coordinator gtest_vendor rosidl_cli
  132 packages not processed
invalid syntax (<unknown>, line 1)
make[1]: *** [Makefile:104: ros2.build] Error 1
make[1]: Leaving directory '/work/pkg/ros2'
make: *** [Makefile:22: all] Error 1
wruser@vxros2:/work$ 
```

mkrunic-windriver, 2023-09-07:
> I have the similar issue with `make` in docker on `humble-release-23.03`, host Ubuntu 22.04:
>
> ```
> --------------------------------------------------------------------------------
> Building [...]

When I remove `cyclonedds` from the build then I get the following error:
```
Building
--------------------------------------------------------------------------------
cd /work/build/ros2/ros2_ws; \
export COLCON_HOME=/work/build/ros2/ros2_ws/.colcon; \
colcon build --mixin vxworks --packages-up-to-regex examples_rclcpp_minimal_timer examples_rclcpp_minimal_client examples_rclcpp_minimal_service examples_rclcpp_minimal_publisher examples_rclpy_* ros2action ros2cli ros2component ros2doctor ros2interface ros2lifecycle ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2topic
/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_devtools/verb/extension_points.py:4: UserWarning: 'colcon_core.entry_point' has been deprecated, use 'colcon_core.extension_point' instead
  from colcon_core.entry_point import EXTENSION_POINT_GROUP_NAME
Starting >>> ament_package
Starting >>> gtest_vendor                     
Starting >>> domain_coordinator
Starting >>> ament_index_python                                                                             
Starting >>> rosidl_cli
Starting >>> fastcdr                                                                                            
Starting >>> rpyutils                                                                                           
Starting >>> osrf_pycommon
--- stderr: domain_coordinator     
```
.
.
.
```
SyntaxError: invalid syntax
---
Failed   <<< rosidl_cli [2.03s, exited with code 1]
Aborted  <<< ament_package [2.84s]                                                                                          
Aborted  <<< fastcdr [2.57s]
Aborted  <<< rpyutils [2.49s]
Aborted  <<< gtest_vendor [2.80s]
Aborted  <<< osrf_pycommon [2.43s]
                                  
Summary: 0 packages finished [9.09s]
  3 packages failed: ament_index_python domain_coordinator rosidl_cli
  5 packages aborted: ament_package fastcdr gtest_vendor osrf_pycommon rpyutils
  4 packages had stderr output: ament_index_python domain_coordinator gtest_vendor rosidl_cli
  131 packages not processed
invalid syntax (<unknown>, line 1)
make[1]: *** [Makefile:104: ros2.build] Error 1
make[1]: Leaving directory '/work/pkg/ros2'
make: *** [Makefile:22: all] Error 1
wruser@vxros2:/work$ 
```

borntocodeRaj, 2023-09-08:
I also ran into the same problem as @mkrunic-windriver 
--------------------------------------------------------------------------------
Building
--------------------------------------------------------------------------------
cd /work/build/ros2/ros2_ws; \
export COLCON_HOME=/work/build/ros2/ros2_ws/.colcon; \
colcon build --mixin vxworks --packages-up-to-regex examples_rclcpp_minimal_timer examples_rclcpp_minimal_client examples_rclcpp_minimal_service examples_rclcpp_minimal_publisher examples_rclpy_* demo_nodes_cpp demo_nodes_py dummy_robot ros2action ros2cli ros2component ros2doctor ros2interface ros2lifecycle ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2topic
/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_devtools/verb/extension_points.py:4: UserWarning: 'colcon_core.entry_point' has been deprecated, use 'colcon_core.extension_point' instead
  from colcon_core.entry_point import EXTENSION_POINT_GROUP_NAME
Starting >>> ament_package
Starting >>> gtest_vendor
Starting >>> domain_coordinator
Starting >>> ament_index_python
Starting >>> rosidl_cli
Starting >>> fastcdr
Starting >>> rpyutils
Starting >>> cyclonedds
Starting >>> osrf_pycommon
Starting >>> urdfdom_headers
Starting >>> google_benchmark_vendor
--- stderr: domain_coordinator
Traceback (most recent call last):
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_ros/package_augmentation/ros_ament_python.py", line 57, in getter
    return get_setup_information(
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 293, in _get_setup_information
    return ast.literal_eval(output)
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/ast.py", line 62, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/work/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 1
    {'package_data': {}, 'dist_files': [], 'src_root': None, 'dependency_links': [], 'setup_requires': [], 'eager_resources': None, 'entry_points': None, 'exclude_package_data': None, 'extras_require': {}, 'include_package_data': None, 'install_requires': ['setuptools'], 'namespace_packages': None, 'packages': ['domain_coordinator'], 'python_requires': None, 'test_loader': None, 'test_runner': None, 'test_suite': None, 'tests_require': ['pytest'], 'use_2to3': None, 'zip_safe': True, 'pbr': None, 'use_scm_version': None, 'verbose': 1, 'dry_run': 0, 'help': 0, 'command_packages': None, 'script_name': 'setup.py', 'script_args': ['--dry-run'], 'command_options': {}, 'package_dir': {}, 'py_modules': None, 'libraries': None, 'headers': None, 'ext_package': None, 'include_dirs': None, 'extra_path': None, 'scripts': None, 'data_files': [('share/domain_coordinator', ['package.xml']), ('share/ament_index/resource_index/packages', ['resource/domain_coordinator'])], 'password': '', 'command_obj': {}, 'have_run': {}, 'want_user_cfg': True, 'metadata': {'name': 'domain_coordinator', 'version': '0.10.0', 'author': 'Pete Baughman', 'author_email': 'pete.baughman@apex.ai', 'maintainer': 'Michel Hidalgo', 'maintainer_email': 'michel@ekumenlabs.com', 'url': 'https://github.com/ros2/ament_cmake_ros', 'license': 'Apache License, Version 2.0', 'description': 'A tool to coordinate unique ROS_DOMAIN_IDs across multiple processes', 'long_description': 'This package provides the functionality used by ament_cmake_pytest_isolated, ament_cmake_gtest_isolated, and launch_testing to select unique ROS_DOMAIN_IDs that allow tests to run in parallel without interfering with each other.', 'keywords': ['ROS'], 'platforms': None, 'classifiers': ['Intended Audience :: Developers', 'License :: OSI Approved :: Apache Software License', 'Programming Language :: Python', 'Topic :: Software Development'], 'download_url': 'https://github.com/ros2/ament_cmake_ros/releases', 'provides': None, 'requires': None, 'obsoletes': None, 'long_description_content_type': None, 'project_urls': {}, 'license_file': None, '_normalized_install_requires': {'setuptools': <Requirement('setuptools')>}, '_normalized_extras_require': {}}}
                                                                                                                                                                                                                                                                                                                  [... cut ...]
  4 packages failed: ament_index_python domain_coordinator osrf_pycommon rosidl_cli
  7 packages aborted: ament_package cyclonedds fastcdr google_benchmark_vendor gtest_vendor rpyutils urdfdom_headers
  5 packages had stderr output: ament_index_python domain_coordinator gtest_vendor osrf_pycommon rosidl_cli
  152 packages not processed
invalid syntax (<unknown>, line 1)
make[1]: *** [Makefile:99: ros2.build] Error 1
make[1]: Leaving directory '/work/pkg/ros2'
make: *** [Makefile:22: all] Error 1

mkrunic-windriver, 2023-09-08:
I have reproduce the same issue on the fresh Ubuntu 22.04 VM instalation, and humble-release-23.03 branch, after running `make` in docker container:
```
--- stderr: domain_coordinator                                                                                                                                                                            
Traceback (most recent call last):
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_ros/package_augmentation/ros_ament_python.py", line 57, in getter
    return get_setup_information(
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 293, in _get_setup_information
    return ast.literal_eval(output)
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/ast.py", line 62, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/wrsdk/vxsdk/host/x86_64-linux/lib/python3.9/ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 1
```

razr, 2023-09-10:
created #88

closed 2023-11-18 by razr