Wind-River/vxworks7-ros2-build #77: Can not exchange messages between two nodes running on differen hosts Ubuntu/VxWorks

mkrunic-windriver, 2023-08-11:
I have setup my environment according to [instructions](https://github.com/Wind-River/vxworks7-ros2-build/blob/master/README.md), but when I try ros2 multicast from both directions nothing is being received on the other side:

https://github.com/ros2/ros2cli/tree/master/ros2multicast

In the Wireshark the message is not detected. When I execute the following line on the VxWorks:
[vxWorks *]# ifconfig gei0 inet add 224.0.0.1/8

message is then captured by Wireshark, but not with ros2 multicast receive.

Just to emphasize that I can ping other side from both directions.

Also I have added this line to vxscript:
taskSpawn 0,50,0,0x20000,setenv,"RMW_IMPLEMENTATION","rmw_cyclonedds_cpp"

and:
export RMW_IMPLEMENTATION="rmw_cyclonedds_cpp"

on Ubuntu, to be sure that I am using the same version of DDS.

razr, 2023-08-14:
Thanks for reporting it. Please do

Host side:

```bash
# add a multicast route to the tap0
$ sudo ip route add 225.0.0.0/8 dev tap0

$ sudo ip route show
...
192.168.200.0/24 dev tap0 proto kernel scope link src 192.168.200.254 
225.0.0.0/8 dev tap0 scope link

$ ros2 multicast receive
Waiting for UDP multicast datagram...
```

VxWorks side:

```bash
[vxWorks *]# python3 ros2 multicast send
Launching process 'python3' ...
Process 'python3' (process Id = 0xffff800000361780) launched.
Sending one UDP multicast datagram...
```

On the host side you should get

```bash
Received from 192.168.200.1:50614: 'Hello World!'
```

mkrunic-windriver, 2023-08-14:
I can confirm that sending and receiving UDP packages using ros2 multicast is now successful.

Now the original problem still persists. I have created [simple ros2 publisher subscriber](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html) packages. When I run publisher and subscriber on the same host (VxWorks or Ubuntu), subscriber successfully receives messages send by publisher, but when subscriber is run on a different host (VxWorks or Ubuntu), messages is not received.     

As I have described above  I have added this line to vxscript:
taskSpawn 0,50,0,0x20000,setenv,"RMW_IMPLEMENTATION","rmw_cyclonedds_cpp"

and this to Ubuntu:
export RMW_IMPLEMENTATION="rmw_cyclonedds_cpp"

just to be sure they are using the same dds middleware.

Is there anything else I have missed to configure to make these two nodes on different hosts to talk to each other?

razr, 2023-08-14:
I'm still digging into the network setting problem.

razr, 2023-08-21:
Starting from ROS 2 `iron` release the `ROS_STATIC_PEER` environment variable has been introduced. it works for the docker/host combination, see https://github.com/eProsima/Fast-DDS/discussions/3811.  `ROS_STATIC_PEER` does not exist in `humble`, you need to use an XML profile instead. eProsima has recommended using [this profile](https://fast-dds.docs.eprosima.com/en/latest/fastdds/discovery/simple.html#initial-peers) to communicate between two peers.

borntocodeRaj, 2023-08-23:
So here are the steps 
On the docker side 
put the xml profile name it as fastdds-docker.xml in vxworks7-ros2-build repo

you can fill the file with the config as 
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="initial_peers_example_profile" is_default_profile="true">
        <rtps>
            <builtin>
                <initialPeersList>
                    <locator>
                        <udpv4>
                            <address>172.17.0.4</address>
                            <port>7412</port>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
        </rtps>
    </participant>
</profiles>

The IP address should be the IP address of the host. 
Export the file in docker 
export FASTRTPS_DEFAULT_PROFILES_FILE=/work/fastdds-docker.xml
Docker should be launches as 
docker run -ti -h ros2native -v $PWD:/work vxros2build:humble
wruser@ros2native:/work$ mkdir -p ros2_native/src && cd ros2_native
wruser@ros2native:/work/ros2_native$ vcs import src < /work/build/ros2/ros2_ws/ros2.repos
wruser@ros2native:/work/ros2_native$ colcon build --merge-install --cmake-force-configure --packages-up-to-regex examples_rcl* ros2action ros2component ros2node ros2pkg ros2service ros2topic ros2cli ros2lifecycle ros2multicast ros2param ros2run demo_* dummy_robot launch --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_TESTING:BOOL=OFF

wruser@ros2native:/work/ros2_native/install$ source setup.bash
wruser@ros2native:/work/ros2_native/install$ ros2 run demo_nodes_py listener

Keep this running in one terminal. 

In another terminal, run from the host.

In the Host side, also keep the xml profile as 
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="initial_peers_example_profile" is_default_profile="true">
        <rtps>
            <builtin>
                <initialPeersList>
                    <locator>
                        <udpv4>
                            <address>172.17.0.4</address>
                            <port>7412</port>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
        </rtps>
    </participant>
</profiles>

Keep the IP of the host, and then source the ros2 as 
source /opt/ros/humble/setup.bash 
source the path of the ros, then from the host also build the package as

colcon build --merge-install --cmake-force-configure --packages-up-to-regex examples_rcl* ros2action ros2component ros2node ros2pkg ros2service ros2topic ros2cli ros2lifecycle ros2multicast ros2param ros2run demo_* dummy_robot launch --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_TESTING:BOOL=OFF

Once the build is successful.
export the xml profile, by exporting the path in the host
export FASTRTPS_DEFAULT_PROFILES_FILE=~/vxworks7-ros2-build/fastdds-docker.xml

Now run on the host side 
ros2 run demo_nodes_cpp talker
[INFO] [1692708772.697120937] [talker]: Publishing: 'Hello World: 1'
[INFO] [1692708773.697069323] [talker]: Publishing: 'Hello World: 2'
[INFO] [1692708774.697118381] [talker]: Publishing: 'Hello World: 3'

You should observe 
wruser@ros2native:/work/ros2_native/install$ ros2 run demo_nodes_py listener
[INFO] [1692708772.697120937] [talker]: Publishing: 'Hello World: 1'
[INFO] [1692708773.697069323] [talker]: Publishing: 'Hello World: 2'
[INFO] [1692708774.697118381] [talker]: Publishing: 'Hello World: 3'

In case the connection breaks, try it again

razr, 2023-08-23:
Hi @borntocodeRaj
what is the difference between your setup and what I have described [here](https://github.com/eProsima/Fast-DDS/discussions/3811#discussioncomment-6791562)

borntocodeRaj, 2023-08-24:
Hello @razr (Andrei) the difference I see is the IP address for me in the xml profile is same for the docker and the host.

razr, 2023-08-24:
I'm glad to see that it works on your side. Could you please put the content of both XML files you are using?
I'm also wondering about this. You run `listener`, but the output messages are from the `talker`.
```bash
wruser@ros2native:/work/ros2_native/install$ ros2 run demo_nodes_py listener
[INFO] [1692708772.697120937] [talker]: Publishing: 'Hello World: 1'
[INFO] [1692708773.697069323] [talker]: Publishing: 'Hello World: 2'
[INFO] [1692708774.697118381] [talker]: Publishing: 'Hello World: 3'
```

borntocodeRaj, 2023-08-24:
Hello @razr actually that is a typo, I am actually getting listener  when I run listener, and talker on the side of the terminal, where I type talker, attached are the actual screenshot of that. 
Communication of talker and Listener 
[image]

XML file content 

[image]

razr, 2023-08-28:
It looks like you have an error opening an XML file. Most probably you still run it with a non-peer-to-peer configuration. XML file shall be loaded on both sides

closed 2023-08-29 by razr

razr, 2023-08-29:
The default gateway (`g=192.168.200.254`) was not set in the VxWorks boot line, Thanks @robwoolley for pointing it out.

```bash
[vxWorks *]# route show

INET route table - vr: 0, table: 254
Destination                     Gateway                       Flags    Use   If    Metric
0.0.0.0/0                       host                          UGS      0     gei0   0
```

mkrunic-windriver, 2023-08-30:
I can confirm that the communication between VxWorks and Host (Ubuntu 22.04.) is working now on my side as well when adding default route. One more thing I had to do:
```
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```
In my case `rmw_cyclonedds_cpp` was set as ros2 middleware, but default is `rmw_fastrtps_cpp`.

Thanks for support @razr and @rcwoolley!

Cheers,
Momo