Wind-River/vxworks7-ros2-build #105: How to generate a CMAKE_TOOLCHAIN_FILE for VxWorks6.9 ？

yuzu-ogura, 2024-04-19:
Hello ,I've been blocked for a month. I want build Fast-DDS repository for VxWorks6.9, but I only have a cross compiler, I don't know how to generate a `CMAKE_TOOLCHAIN_FILE`. 
Besides, it seems like Linux have no cross compiler for VxWorks6.9. Because such as `./configure`、`automake`、`autoconf`，such tools is inconvenient for Windows, how to build an cross compiler for VxWorks6.9, I try a lot but it's useless.
Pls help me how to solve this two hard problems.

razr, 2024-04-19:
Hi @yuzu-ogura we do provide a cross-compiler together with VxWorks 6.9 for both Linux and Windows hosts. The best way to solve your problem is to contact your local Wind River support team. Personally, I have never used VxWorks 6.9 and CMake. @kuhlenough do we have any documentation on using CMake under VxWorks 6.9?

kuhlenough, 2024-04-19:
No, any CMake support Wind River has done has been for recent VxWorks 7 versions.  As part of VSB (VxWorks Source Build) all the CMakes stubs are created for your toolchain and BSP.  VxWorks 6.9 is EOL originally release is 2010 when Windows 7 and Ubuntu 10 were current, would you expect them to support FastDDS ?

closed 2024-05-07 by yuzu-ogura