Wind-River/vxworks7-ros2-build #106: What's the password of default user "wruser"?

yuzu-ogura, 2024-04-30:
Hello, I want modify the file owner and owner group. It's really funny...
[image]

razr, 2024-04-30:
password is `wruser`. However, it does not help you because `wruser` is not in the `sudo`.
Could you please provide the output of your `id` and `ls -la` commands while inside your docker. I'm wondering what user id you have versus your filesystem. It looks like they are not the same.

```bash
docker run -ti -h vxros2 -e UID=2000 -e GID=2000 -v ~/Downloads/wrsdk:/wrsdk -v $PWD:/work vxros2build:humble

wruser@vxros2:/work$ id
uid=2000(wruser) gid=2000(wruser) groups=2000(wruser)

wruser@vxros2:/work$ ls -la
total 120
drwxrwxr-x 16 1000 1000  4096 Apr 29 09:28 .
drwxr-xr-x  1 root root  4096 Apr 30 08:17 ..
drwxrwxr-x  8 1000 1000  4096 Apr 29 08:17 .git
drwxrwxr-x  3 1000 1000  4096 Aug  5  2022 .github
-rw-rw-r--  1 1000 1000     8 Feb 10 18:54 .gitignore
```

yuzu-ogura, 2024-04-30:
@razr Thanks for you quick reply. pls seen below:
[image]
It seems thats they are the same. That's really puzzled me a lot

yuzu-ogura, 2024-04-30:
Thanks. The `/wrsdk` is in my `HOME` directory, Its owner and group like below, is this wrong?
[image]

yuzu-ogura, 2024-04-30:
> your `/wrsdk` has probably `root:root` permissions. you need to change it to `wruser:wruser`. I don't think you should do `chmod -R 777`. It should work properly without it

Is your mean that I should change the permission of `/wrsdk` to `wruser:wruser` in the Docker container?

razr, 2024-04-30:
it is fine, you just need to restore permissions of your `/work` directory to `liuyu:liuyu` and run docker with ````-e UID=`id -u`-e GID=`id -g```` as

```bash
docker run -ti -h vxros2 -e UID=`id -u`-e GID=`id -g`-v ~/Downloads/wrsdk:/wrsdk -v $PWD:/work vxros2build:humble
```

After that, everything should work. Inside your docker container your `id` and filesystem `owner:group` shall be the same for `/work` and `/wrsdk`

yuzu-ogura, 2024-04-30:
Tell the truth, the aboving error log is produced by the following command. I've confirmed it many times :
`docker run -ti -h vxros2 -e UID=`id -u` -e GID=`id -g` -v ~/Downloads/wrsdk:/wrsdk -v $PWD:/work vxros2build:humble'
Here are my questions:
1、In my pc file permission is 'liuyu:liuyu' but in the docker container it changes to 'root:root'
2、I seen these ENV are fixed:
[image]

razr, 2024-04-30:
do you run docker inside Ubuntu on a Windows WSL?

yuzu-ogura, 2024-04-30:
> do you run docker inside Ubuntu on a Windows WSL?

I install Ubuntu on the `VMWare` virtual machine. And  `VMWare` install on `Windows 11`

razr, 2024-04-30:
I don't know much about this combination, but I remember that the `docker volume/Ubuntu/WSL/Windows` didn't work properly. I assume it is the same with VMware.
Try to run 
`docker run -ti -h vxros2 -e USER=root -v ~/Downloads/wrsdk:/wrsdk -v $PWD:/work vxros2build:humble`
However, I didn't try whether the build works as `root`

yuzu-ogura, 2024-04-30:
> I don't know much about this combination, but I remember that the `docker volume/Ubuntu/WSL/Windows` didn't work properly. I assume it is the same with VMware. Try to run `docker run -ti -h vxros2 [...]

Thank you very much. It seems I run this command successfull. By the way, execution stops here , Is my network bad?
[image]

razr, 2024-04-30:
I'm not sure what you mean by "stops". Build should fail in the end if it was not possible to download repos.

yuzu-ogura, 2024-05-07:
> Hello, I want modify the file owner and owner group. It's really funny... ![image](https://private-user-images.githubusercontent.com/166282486/326657785-d92e1d33-2787-4fc7-9ca4-43917ba1d32b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwNTQ0OTIsIm5iZiI6MTcxNTA1NDE5MiwicGF0aCI6Ii8xNjYyODI0ODYvMzI2NjU3Nzg1LWQ5MmUxZDMzLTI3ODctNGZjNy05Y2E0LTQzOTE3YmExZDMyYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QwMzU2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hN2EzMzE2ODBhMzQ3ZjBmNTgxNmI3Zjg5OTU0NzQ2OGQxMGJhMjE4Y2EyM2Q0NTkyZTMxYTI3OTRiMjY2OWM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fPwzkg9Ahfg452xKIR945dkW-GrS-LNbQPyQm_kO8iU) After I Run Docker image, I'm going to make , but occuered an error (permission denied), like below: ![image](https://private-user-images.githubusercontent.com/166282486/326658193-73c256ff-fbf6-4011-aa37-e717e67e1af1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwNTQ0OTIsIm5iZiI6MTcxNTA1NDE5MiwicGF0aCI6Ii8xNjYyODI0ODYvMzI2NjU4MTkzLTczYzI1NmZmLWZiZjYtNDAxMS1hYTM3LWU3MTdlNjdlMWFmMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QwMzU2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODczOGU2NTBkNzk0ZDJkNTcwMmU2ODcxODllMGNmMDUwODVlYjJlMjQ3OTgwODNjZjNhZTg5ZWRhMzlkNzE4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9._Yq-kWeGET2APzdWmt_Otyq7NMM7YrkRVm2q_yJtl78)
> 
> another error： ![image](https://private-user-images.githubusercontent.com/166282486/326692331-75a3c09d-9952-4aae-be8f-ae4dd8429829.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwNTQ0OTIsIm5iZiI6MTcxNTA1NDE5MiwicGF0aCI6Ii8xNjYyODI0ODYvMzI2NjkyMzMxLTc1YTNjMDlkLTk5NTItNGFhZS1iZThmLWFlNGRkODQyOTgyOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QwMzU2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNWUzNWVjNmYyMWE4MjUzMmE1N2Q1NjhmYzYwMDdkOTJhY2JiYWRmZjcwMmIwZjBkNWZjMWM4MDQ4OWZlMDk2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2RWuO2BDsiX1dB7O675WYgAAfopZSfrVYKnYbFeGOJQ) so, I try to modify the file owner and owner group : ![image](https://private-user-images.githubusercontent.com/166282486/326656378-a3b00440-8f1c-498e-ad46-f13061ccc125.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwNTQ0OTIsIm5iZiI6MTcxNTA1NDE5MiwicGF0aCI6Ii8xNjYyODI0ODYvMzI2NjU2Mzc4LWEzYjAwNDQwLThmMWMtNDk4ZS1hZDQ2LWYxMzA2MWNjYzEyNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QwMzU2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kN2U2ZDQyMjViYzY3ZTBmYTZjOWY5ODI0MmFhOGY4N2U2MDQxOGE0OWQwMDVkODI2MTY2Y2M2NGM1ZDlmNDY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.u8VivOPvVXSPVWBnv95UVj399Q19WMiNdHpPTEQ-xFw)

Thank you for your patient answer. After I uninstall the docker and reinstall the docker, the problem about `permission denied` never happened again.

closed 2024-05-07 by yuzu-ogura