Wind-River/vxworks7-ros2-build #108: Resolves signal 11 error issues

0312ALEX, 2024-05-28:
Hello, @razr when I run this program in the Ubuntu virtual machine, there is a signal 11 error. I do not know how to modify it. 
[image]
And not only this statement will give this error, other simple statements such as "python3" will also give this error. There are too few related information on the Internet, I can't solve the problem by myself.I hope you can provide some help
looking forward to your reply

razr, 2024-06-07:
sorry for replying late, I was away. What target do you run? Is it QEMU or Raspberry?

0312ALEX, 2024-06-12:
that's all right@razr. it's QEMU ,I'm running on QEMU

0312ALEX, 2024-06-12:
it's okay@razr. it's QEMU ,I'm running on QEMU