Wind-River/vxworks7-ros2-build #112: how load ros2.img without qemu ？

zhang125, 2025-02-19:


razr, 2025-02-20:
Duplicate of #111

zhang125, 2025-02-27:
Thank you for your reply.  However, I have successfully run the ros2-demo.
My hardware platform is based on an i7 processor with a Kaylike architecture. The Workbench version is 23.09. I can successfully boot the VxWorks image from a USB drive and also successfully load the ros2 script files.
However, when I run the ros2-demo, it prompts an error:
[vxWorks *]# 
**[vxWorks *]# ../lib/examples_rclcpp_minimal_timer/timer_lambda
Fatal error**
[vxWorks *]#

On Ubuntu 22.04, the ros2 compilation environment is ros2-humble-24.03, and QEMU_x86_64 is 24.03. I can successfully run the ros2-demo in QEMU.
I formatted the USB drive as FAT32, created the EFI/BOOT directory, copied the files from qemu's sdcat/sysroot to the EFI directory, and also copied the files from the ros.img to the EFI directory. However, even after doing all this, I still can't run it. Where did I go wrong in my operations?

-> pwd
/bd0a/EFI/ROS2/bin
value = 8928351037065411072 = 0x7be7dff6ba463a00
-> 
-> 
-> 
-> ls "/bd0a/EFI/ROS2"
/bd0a/EFI/ROS2/. 
/bd0a/EFI/ROS2/.. 
/bd0a/EFI/ROS2/VXSCRIPT 
/bd0a/EFI/ROS2/BIN 
/bd0a/EFI/ROS2/LIB 
/bd0a/EFI/ROS2/SHARE 
/bd0a/EFI/ROS2/VXSCRI~1 
/bd0a/EFI/ROS2/vxscript_ros2 
value = 0 = 0x0
/bd0a/EFI
[vxWorks *]# 
[vxWorks *]# ls
. 
.. 
BOOT 
sysroot 
ROS2 
[vxWorks *]#

razr, 2025-02-28:
You make a good progress! Please consider:
1. VxWorks is a case sensitive, use low case for all directories, files etc. It should be `ros2/vxscript` instead of `ROS2/VXSCRIPT`
2. If you have `23.09` VxWorks it is better to compile ROS2 against `23.09`, not `24.03`
3. Please make sure that you have built your VSB and VIP with configuration defined in the SDK:
 - `vxsdk/sysroot/vsb_config.json`
 - `vxsdk/bsps/itl_generic_3_0_0_4/vip_config.json`
4. there are settings in the `vxscript` which you need to apply prior to run ROS2

```bash
symlink "/bd0a","/usr"
shellConfigDefaultValueSet "VXE_PATH", "/usr/bin"
rtpSpStackSize=0x100000
setenv,"LD_LIBRARY_PATH","/usr/lib"
setenv,"AMENT_PREFIX_PATH","/usr"
"ROS_LOG_DIR","/ram0"
```

zhang125, 2025-03-01:
> You make a good progress! Please consider:
> 
> 1. VxWorks is a case sensitive, use low case for all directories, files etc. It should be `ros2/vxscript` instead of `ROS2/VXSCRIPT`
> 2. If you have `23.09` VxWorks it is better to compile ROS2 against `23.09`, not `24.03`
> 3. Please make sure that you have built your VSB and VIP with configuration defined in the SDK:
> 
> * `vxsdk/sysroot/vsb_config.json`
> * `vxsdk/bsps/itl_generic_3_0_0_4/vip_config.json`
> 
> 4. there are settings in the `vxscript` which you need to apply prior to run ROS2
> 
> symlink "/bd0a","/usr"
> shellConfigDefaultValueSet "VXE_PATH", "/usr/bin"
> rtpSpStackSize=0x100000
> setenv,"LD_LIBRARY_PATH","/usr/lib"
> setenv,"AMENT_PREFIX_PATH","/usr"
> "ROS_LOG_DIR","/ram0"

Thanks a lot,l learned very mush.
I checked    vsb&vip, changed the script, but it not work. so i try use 23.09 build the ros2,but it has a error, so can you help me again?   24.03 is ok.

--- stderr: google_benchmark_vendor                                                 
CMake Error at /wrsdk/vxsdk/host/x86_64-linux/share/cmake-3.26/Modules/CMakeTestCXXCompiler.cmake:60 (message):
  The C++ compiler 
    "/wrsdk/vxsdk/host/x86_64-linux/bin/wr-c++"   is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A
    
    Run Build Command(s):/wrsdk/vxsdk/host/x86_64-linux/bin/cmake -E env VERBOSE=1 /wrsdk/vxsdk/host/x86_64-linux/bin/make -f Makefile cmTC_1ab42/fast && make[2]: Entering directory '/work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A'
    /wrsdk/vxsdk/host/x86_64-linux/bin/make  -f CMakeFiles/cmTC_1ab42.dir/build.make CMakeFiles/cmTC_1ab42.dir/build
    make[3]: Entering directory '/work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A'
    Building CXX object CMakeFiles/cmTC_1ab42.dir/testCXXCompiler.cxx.o
    /wrsdk/vxsdk/host/x86_64-linux/bin/wr-c++ --sysroot=/wrsdk/vxsdk/sysroot   -fwarning-level-low  -MD -MT CMakeFiles/cmTC_1ab42.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_1ab42.dir/testCXXCompiler.cxx.o.d -o CMakeFiles/cmTC_1ab42.dir/testCXXCompiler.cxx.o -c /work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A/testCXXCompiler.cxx
    clang++: error: unknown argument: '-fwarning-level-low'
    make[3]: *** [CMakeFiles/cmTC_1ab42.dir/build.make:79: CMakeFiles/cmTC_1ab42.dir/testCXXCompiler.cxx.o] Error 1
    make[3]: Leaving directory '/work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A'
    make[2]: *** [Makefile:127: cmTC_1ab42/fast] Error 2
    make[2]: Leaving directory '/work/output/build/ros2/ros2_ws/build/google_benchmark_vendor/CMakeFiles/CMakeScratch/TryCompile-zhkb2A'

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)
---
Failed   <<< google_benchmark_vendor [1.15s, exited with code 1]

razr, 2025-03-04:
I'll look into it. The build is set with `-DBUILD_TESTING=ON` and `23.09` does not work with it.

razr, 2025-03-10:
please try it now.