Wind-River/vxworks7-ros2-build #115: Calling the VXWORKS API in ROS2 code, compilation error

zhang125, 2025-07-04:
Hi, When I use ros2, I need to call the VxWorks api, but the compilation always fails, is it the wrong method? If you don't call the system function, there's no problem. Here's the code:
### **Error:**
/wrsdk/vxsdk/host/x86_64-linux/bin/../../../sysroot/usr/h/public/base/b_pthread_mutex_t.h:40:5: error: unknown type name '_Vx_SEM_ID'
   40 |     _Vx_SEM_ID          mutexSemId;   /* Vx semaphore associated with */
      |     ^
/wrsdk/vxsdk/host/x86_64-linux/bin/../../../sysroot/usr/h/public/base/b_pthread_cond_t.h:41:5: error: unknown type name '_Vx_SEM_ID'
   41 |     _Vx_SEM_ID          condSemId;      /* Vx sem associated with */ 
      |     ^
/wrsdk/vxsdk/sysroot/krnl/h/public/base/b_trace_id_t.h:26:9: error: unknown type name '_Vx_OBJ_HANDLE'; did you mean 'OBJ_HANDLE'?
   26 | typedef _Vx_OBJ_HANDLE  trace_id_t;

### **Code:**
//#include <iostream>

extern "C"{
#include <vxWorks.h>
#include <semLib.h>
#include <pthread.h>
#include <sysLib.h>
#include <taskLib.h>
}

// 其次包含 C++ 标准库（避免冲突）
//#include <cstdio>  // 优先使用 C 风格的 I/O
#include <iostream>
using namespace std;

// 任务入口函数，必须是C风格
extern "C" int myTask(int arg)
{
    for (int i = 0; i < 5; ++i)
    {
        cout << "Hello from myTask, arg=" << arg << ", i=" << i << endl;
        taskDelay(60); // 延时60个tick
    }
    return 0;
}

int main_vxworks()
{
    // 创建任务
    TASK_ID tid = taskSpawn(
        (char *)"tMyTask",   // 任务名
        100,                 // 优先级
        0,                   // 选项
        0x1000,              // 堆栈大小
        (FUNCPTR)myTask,     // 入口函数
        123,0,0,0,0,0,0,0,0,0 // 参数，最多10个
    );

    if (tid == NULL)
        cout << "taskSpawn failed!" << endl;
    else
        cout << "taskSpawn success, tid=" << tid << endl;

    // 主任务延时，等待子任务输出
    taskDelay(300);

    return 0;
}

int main(int argc, char * argv[])
{ 
        for(int i=0;i<100;i++)
        cout << "Hello World!" << endl; 

        main_vxworks(); 
        return 0;
}
### **CMakeList.txt:**

vxros2@vxros2-Default-string:~/vxworks7-ros2-build/output/build/ros2/ros2_ws/src/my_package_vxworks1$ cat CMakeLists.txt 
cmake_minimum_required(VERSION 3.8)
project(my_package_vxworks1)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

include_directories(
    /wrsdk/vxsdk/sysroot/krnl/h
    /wrsdk/vxsdk/sysroot/krnl/h/public
)

add_executable(my_package_vxworks1 src/my_package_vxworks1.cpp)
install(TARGETS
  my_package_vxworks1
  DESTINATION lib/${PROJECT_NAME}
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # comment the line when a copyright and license is added to all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # comment the line when this package is in a git repo and when
  # a copyright and license is added to all source files
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()

zhang125, 2025-07-06:
> your build command is correct. Please use user-space APIs (RTP, POSIX etc.)

Thank you very much for your reply, I solved the interaction between ros2 and memory by sharing memory.   
  but I encountered a new error prompt, DDS does not support shm transfer, this is when compiling, is there any macro definition?

[vxWorks *]# /usr/lib/examples_rclcpp_minimal_publisher/publisher_lambda
2025-07-06 10:18:32.756 [RTPS_MSG_OUT Error] error: couldn't generate segment_name -> Function init
2025-07-06 10:18:32.756 [RTPS_PARTICIPANT Error] Unable to Register SHM Transport. SHM Transport is not supported in the current platform. -> Function RTPSParticipantImpl

razr, 2025-07-06:
what VxWorks version are you using? I have added [a shared memory support](https://github.com/Wind-River/vxworks7-layer-for-ros2/blob/humble-release-24.03/Fast-DDS/usr_src/0011-use-shared-memory-tmp-and-_RDWR-for-LOCK_EX.patch) for the `24.03`

zhang125, 2025-07-06:
my vxworks version is 23.09，maybe support??发自我的手机-------- 原始邮件 --------发件人： Andrei Kholodnyi ***@***.***>日期： 2025年7月6日周日 21:45收件人： Wind-River/vxworks7-ros2-build ***@***.***>抄送： zhang125 ***@***.***>, Author ***@***.***>主    题： Re: [Wind-River/vxworks7-ros2-build] Calling the VXWORKS API in ROS2 code, compilation error (Issue #115)razr left a comment (Wind-River/vxworks7-ros2-build#115)
what VxWorks version are you using? I have added a shared memory support for the 24.03

zhang125, 2025-07-08:
thanks，23.09  version devs  has /shm，i will try  it.发自我的手机-------- 原始邮件 --------发件人： Andrei Kholodnyi ***@***.***>日期： 2025年7月7日周一 16:35收件人： Wind-River/vxworks7-ros2-build ***@***.***>抄送： zhang125 ***@***.***>, Author ***@***.***>主    题： Re: [Wind-River/vxworks7-ros2-build] Calling the VXWORKS API in ROS2 code, compilation error (Issue #115)razr left a comment (Wind-River/vxworks7-ros2-build#115)
do you see /shm under devs in 23.09? Then shared memory is supported. If you want Fast-DDS to support shared memory as well, just apply the patch I mentioned above.
-> devs
drv refs name
  5 [ 5] /ata0
  5 [ 5] /bd0a
 13 [ 5] /dev/random
 13 [ 5] /dev/urandom
 12 [ 5] /dev/zero
 11 [ 5] /fifos
 16 [ 5] /host.host
 14 [ 5] /input/event
  0 [ 7] /null
  5 [ 5] /ram
 10 [ 5] /ram0
  9 [ 5] /romfs
  3 [ 5] /shm